RoelKroes/TBTracker-ESP32 #4: LoRa-APRS packets should include a timestamp

darksidelemm, 2024-07-14:
Currently you are sending APRS packets that do not use a timestamp.

This means that any software interpreting these packets is forced to use the 'arrival time' of the packet as its timestamp. Usually this is OK, as the APRS-IS network should de-duplicate packets. However, many LoRa APRS igates are adding their own metadata (rssi, snr) onto the comment field of packets, which completely breaks any de-duplication.

This means that software processing data sees *multiple* packets of the same position which different timestamps. This breaks things like estimates of ascent rate and ground speed.

If you at least use a packet format that includes a timestamp (e.g. the HHMMSSh format) then software can use this information to handle duplicate packets in a better way (they will at least have the same timestamp!).

RoelKroes, 2024-08-28:
I added a timestamp to the APRS packets. 

Packets will now look like:

PD7R-11>APZTBT:@280240z5213.07N/00509.78EO/A=000022 LoRa-APRS HAB-NL

So source, dest-id, position report with timestamp (ddHHMM zulu time) and a comment field with altitude and free user text.

If you have more recommendations please let me know. I really appreciate it. Otherwise I will add this to the repository.

Thank you for your help!

Roel.

darksidelemm, 2024-08-28:
Use HHMMSS, not ddHHMM.

By using ddHHMM, you have now:
- Limited transmissions to only once per minute.
- Quantized the timestamp to the nearest minute, which could do very weird things to ascent rate and speed calculations.
- 
We can handle the lack of date in the packet, we already do that with Horus Binary telemetry anyway.

RoelKroes, 2024-08-28:
Hi Mark,

Okay makes Sense. I changed it.
The packet will look like: PD7R-11>APZTBT:@224227h5213.07N/00509.78EO/A=000072 LoRa-APRS HAB-NL

Is that okay for Sondehub?

darksidelemm, 2024-08-28:
That should be fine! It looks like they are showing up OK.

RoelKroes, 2024-08-30:
I've added the code for device-id and timestamp to the repository and created a new release.
Closing this issue, thanks for your help!

closed 2024-08-30 by RoelKroes