RoelKroes/TBTracker-ESP32 #5: Option for uplink?

Mika-1818, 2024-12-08:
I'd be interested in uplink(e.g for cutdown) and have a feeling that I won't be able to modify your code in a way it works reliably.
I think it would be a really cool and useful feature, especially because there aren't many payloads that support this.
It could be implemented like the voltage and temperature transmission so one can write it's own code for cutdown.
If you could integrate that, that'd be awesome!
vy73 de Mika, DL8MIK

RoelKroes, 2024-12-10:
Hi Mika,

Thanks for your suggestion. That is indeed a good suggestion.
I will try to implement this but it will take some time. 
I will leave this thread open. If you (or anybody) have additional info/tips for this option, just add to this thread.

Thanks,

Roel.

Mika-1818, 2024-12-10:
That's great! I'd be happy to do a few test-launches. 
One Idea might be to also modify the tbtracker-rx to send uplink. Timing could be done by listening to the downlink and sending after that has finished.

zanco, 2024-12-21:
So, my 2 cents ?
Uplink to TBTrackerTX sounds nice, if there is indeed need for cutdown or switching something on or off. I would suggest to use a different frequency for the uplink to prevent other guys to cutdown your payload. I have seen cutdown systems with a piece of resistor wire burning the balloon wire by the way.

I like Mika's idea to have tbtracker-rx send something back to a tx-ing tbtracker althoug I can not yet find a use for that. But we will invent some use for it :-)

Mika-1818, 2024-12-21:
I think in combination with Wenet it could be really fun to control a servo that is attached to the elevation-axis of the camera. 
I think there are many ways uplink could be used!

pc4l-hab, 2024-12-22:
can we have a digipeater function like ISS :)

Mika-1818, 2024-12-22:
Possible? Probably. Worth it? I don't think so. If you really want it to be a digipeater, I'd go for something like Meshtastic.

RoelKroes, 2025-02-22:
Hi guys, I implemented the uplink function in TBTracker-esp32. It is in the main source tree and in the new v0.2.0 release.

- There is an extra entry in settings.h. For the uplink to work, you need to set LORA_ENABLED and RECEIVING_ENABLED to true.

-  For now the uplink/rx works on the same LoRa parameters as TX. This is maybe not ideal but it is still experimental

- There is a new unit called Receive.ino. This unit contains a function "ProcessRXPacket" that will run each time a packets was received. In that function you can check the received packet and execute some code. I provided an example in the code.

Thank you for the ideas! I hope this will improve the tracker a bit. 
Please, feel free to test it.

Roel.

Mika-1818, 2025-02-22:
Thank you so much for implementing this!
I will try it out soon though I still have to wait on some parts for my custom Tracker pcb.

Mika-1818, 2025-03-22:
Works perfectly fine. I used a modified version of the example code as a transmitter.

closed 2025-03-22 by Mika-1818