RoelKroes/TBTracker-ESP32 #7: 'ReadVCC' was not declared in this scope

justtestjusttestjusttest, 2025-09-01:
I got these warnings when I tried to check the draft:

\Documents\Arduino\TBTracker-ESP32-main\TBTracker-ESP32-main\TBTracker-ESP32\TBTracker-ESP32.ino:138:37: error: 'ReadVCC' was not declared in this scope
  138 |   BinaryPacket.BattVoltage = round((ReadVCC() / 5.0) * 256.0);

\Documents\Arduino\TBTracker-ESP32-main\TBTracker-ESP32-main\TBTracker-ESP32\TBTracker-ESP32.ino:140:37: error: 'ReadTemp' was not declared in this scope
  140 |   BinaryPacket.Temp = (int8_t)round(ReadTemp());

and a few others.
Sorry for the stupid question, as I am a novice, but how can I determine what the problem is?
I thought it would be a plug-and-play solution, but as usual, I have a problem. (Of course, this is my problem and lack of knowledge).

Great job, by the way! Thanks.

RoelKroes, 2025-09-03:
Hi,

Are you sure that you have all files from the code downloaded? And that these files are all in the same folder?
The errors you posted seem to happen because the file ADC.ini is not present in your main folder.

Best,

Roel.

RoelKroes, 2025-09-04:
You can change the TX power directly in settings.h.

There are three settings. One for LoRa, one for Horus and one for AFSK.

#define LORA_POWER 13                    // in dBm between 2 and 17. 10 = 10mW (recommended)
#define HORUS_POWER 13                 // In dBm. Valid values +2 to +17 dBm. 10dBm = 10mW, 13dBm=20mW (recommended)
#define APRS_AFSK_POWER 13           // In dBm. Valid values +2 to +17 dBm. 10dBm = 10mW, 13dBm=20mW

Just lookup those settings in the settings.h file and change them. Then recompile and re-upload.

RoelKroes, 2025-09-09:
Nice to hear it works for you!

Don't worry about supporting my work. Jusy pay it forward. Help or teach somebody else how to install and work with this software (or any other software). 

Best,

Roel.

closed 2025-09-09 by RoelKroes